STMicroelectronics/cmsis-device-h7 #2: Wrong comment for VECT_TAB_OFFSET

carlo-dev-git, 2021-12-15:
This value must be a multiple of 0x200 (generic template)

https://github.com/STMicroelectronics/cmsis_device_h7/blob/f5e3a1d5e71e6b9278a6d580b7d5ca82c0debd9b/Source/Templates/system_stm32h7xx.c#L98

 or 0x300 (other files) ?
https://github.com/STMicroelectronics/cmsis_device_h7/blob/f5e3a1d5e71e6b9278a6d580b7d5ca82c0debd9b/Source/Templates/system_stm32h7xx_dualcore_boot_cm4_cm7.c#L87

ALABSTM, 2022-01-05:
ST Internal Reference: 120332

carlo-dev-git, 2022-03-10:
Hi ALABSTM,

I just updated the STM32Cube STM32H7 MCU Package to v1.10.0 but this ticket is not fixed because it was scheduled to v1.11.0. From stm32cube-mcu-cmsis-dashboard the ticket status is set To release, so could you just tell me what is the correct offset (0x200 or 0x300)?
It is very important for me.

Best regards.

ALABSTM, 2022-03-14:
Hi @carlo-dev-git,

My bad. I mistook the milestone which is `v1.10.2` and not `v1.11.0`. Actually, the typo has been corrected as you can see from the code snippet below. The correct offset value is `0x300`. The fix has been published on both  this repo and [STM32CubeH7](https://github.com/STMicroelectronics/STM32CubeH7).

https://github.com/STMicroelectronics/cmsis_device_h7/blob/a95954cd368b94e60d3dc42b8005a0460a18d7cb/Source/Templates/system_stm32h7xx.c#L98-L99

With regards,

carlo-dev-git, 2022-03-15:
Hi ALABSTM,

thanks for your clarification but what about the single-core template?

https://github.com/STMicroelectronics/cmsis_device_h7/blob/a95954cd368b94e60d3dc42b8005a0460a18d7cb/Source/Templates/system_stm32h7xx_singlecore.c#L96-L106

carlo-dev-git, 2022-03-15:
Hi ALABSTM,
please check **PM0253** _STM32F7 Series and STM32H7 Series Cortex®-M7 processor programming manual_ (Rev 5)

**4.3.4 Vector table offset register**
The table alignment requirements mean that **bits [8:0]** of the table offset are always **zero**.

`0x300 -> 001100000000b`

closed 2023-02-23 by TOUNSTM