STMicroelectronics/cmsis-device-h7 #4: Allow VECT_TAB_OFFSET to be set externally

armandas, 2022-08-12:
Currently, in order to change the value of VECT_TAB_OFFSET, the user has to modify the `system_stm32h7xx.c` file. This is inconvenient in cases where the value has to change for different builds.

If the code was changed from:

```c
#define VECT_TAB_OFFSET         0x00000000U
```

to 

```c
#ifndef VECT_TAB_OFFSET
#define VECT_TAB_OFFSET         0x00000000U
#endif
```

The user could define the value via a command line, like so `-DVECT_TAB_OFFSET=0x00020000`.

utkarshsethi, 2022-09-14:
@ALABSTM I think this is duplicate/would also fix https://github.com/STMicroelectronics/cmsis_device_g4/issues/1

okz, 2023-10-18:
not being able to define this externally, or having this inside a user block, is a real inconvenience, especially when developing seperate boot / application firmware.   

It's are really small fix, is there a reason for not implementing it yet?

ALABSTM, 2023-11-29:
Hi everyone,

@armandas, thank you for having reported. We already logged the point internally and are working on it, among many other points.

@utkarshsethi, you're absolutely right. This is the same issue as the one you mentioned. However, as the repos are different, we will keep track of them separately.

@okz, I understand your point. We will try to have this fix implemented, deployed, and released soon enough hopefully.

Thank you all for your patience and your comprehension.

With regards,

ALABSTM, 2025-02-21:
Fixed in commit 6c7cb3ea1e3f139a1a9e1a7bb13c7ce5540bf76b

ALABSTM, 2025-02-21:
Hi everyone,

I hope you are doing well. We finally updated the [system_stm32h7xx.c](https://github.com/STMicroelectronics/cmsis-device-h7/blob/6c7cb3ea1e3f139a1a9e1a7bb13c7ce5540bf76b/Source/Templates/system_stm32h7xx.c#L120-L123) files to allow macro `VECT_TAB_OFFSET` to be overridden externally.

Thank you for your patience and thank you again for having reported.

With regards,