STMicroelectronics/cmsis-device-h7 #5: Incorrect __CM7_REV and some missing values in device headers

billybednar, 2023-08-12:
**Describe the set-up**
I'm reporting this for the STM32H742/743/750/753 family as I am most familiar with them, but it looks like this applies to all other H7 devices.

**Describe the bug**
The `__CM7_REV` macro indicates that the revision of the M7 core is r1p0, but it's actually r1p1 according to the [errata](https://www.st.com/resource/en/errata_sheet/es0392-stm32h742x743xig-st32h750xb-and-stm32h753xi-device-errata-stmicroelectronics.pdf#page=8) and the CPUID register on a physical device (0x41**1**FC27**1**). I imagine it's a carry over from older F7 devices. Making things more confusing, the datasheet and reference manual do not list the revision and old versions of the errata had a different incorrect value listed (r0p1).

https://github.com/STMicroelectronics/cmsis_device_h7/blob/v1.10.3/Include/stm32h742xx.h#L213

The headers also seem to be missing a few definitions (`__VTOR_PRESENT`, `__FPU_DP`, and `__DTCM_PRESENT`) that are present in the template from the included version of CMSIS.

https://github.com/ARM-software/CMSIS_5/blob/5.6.0/Device/_Template_Vendor/Vendor/Device/Include/Device.h#L101-L111

TOUNSTM, 2023-11-10:
ST Internal Reference: 166097

TOUNSTM, 2024-02-26:
Fixed in commit 3870629f9e39d43bf581df3ff739f06f1546f08f

closed 2024-02-26 by TOUNSTM