STMicroelectronics/cmsis-device-h7 #9: Incorrect RTC register

khezami, 2024-01-12:
**Describe the set-up:**

- I'm reporting this for the STM32H723/733, STM32H725/735 and STM32H730 family

**Describe the bug:**

- The RTC driver for the STM32H723 is incorrect.

- The RTC_TAFCR register, present in STM32H723/733, STM32H725/735 and STM32H730 family devices, is replaced with the RTC_TAMPCR register in the STM32H7xx.

- This discrepancy raises concerns about potential changes in RTC functionality.

**Screenshots:**

_- RM0468 : STM32H723/733, STM32H725/735 and STM32H730 :_

[image]

_- RM0433 : STM32H742, STM32H743/753 and STM32H750 :_ 

[image]

_[- cmsis_device_h7/Include/stm32h723xx.h](https://github.com/STMicroelectronics/cmsis_device_h7/blob/master/Include/stm32h723xx.h)_

[image]

ALABSTM, 2024-01-17:
Hi @khezami,

Thank you for having reported. At first glance, this looks rather as an issue in the documentation. It has been forwarded to the concerned teams for confirmation. I will keep you updated. Thank you again.

With regards,

ALABSTM, 2024-01-17:
ST Internal Reference: 171062

ALABSTM, 2024-11-28:
Hi @khezami,

Back to this point you reported. After further analysis, this is rather related to the definitions in the CMSIS header files. This will be fixed soon enough hopefully. Thank you again for the heads-up.

With regards,

ALABSTM, 2025-02-21:
Hi @khezami,

I hope you are fine. Issue fixed, as you can see from [this](https://github.com/STMicroelectronics/cmsis-device-h7/blob/ee5d51cb07c70a8e631762e159b32f7450602740/Include/stm32h723xx.h#L16976-L17106) sample code. Thank you for your patience and thank you again for having reported.

With regards,

closed 2025-02-21 by ALABSTM