STMicroelectronics/cmsis-device-h7 #10: Reset_Handler resets variables set by SystemInit()

cpq, 2024-01-25:
The very detailed report was done for https://github.com/STMicroelectronics/cmsis_device_l4/issues/2
This issue is not present on some lines, e.g. F4. And present on the majority of others, like L4, G0, H7, etc.
Please fix the order of initialisation - data/bss section first, then SystemInit(), second.

TOUNSTM, 2024-07-19:
Hello @cpq,

As mentioned in the previous issue https://github.com/STMicroelectronics/cmsis_device_l4/issues/2, the main reason for this change is the use of external memories. When using external memory, it is crucial to initialize the memory controller before accessing any variables stored in that memory. This is why SystemInit() must be called before initializing the data and BSS sections.

In the SystemInit() API, there are no variables like SystemCoreClock; instead, it directly accesses hardware registers to configure the system. We have thoroughly tested this approach and have not encountered any cases where SystemCoreClock is equal to 0.
https://github.com/STMicroelectronics/cmsis_device_h7/blob/13ba9da797a30e1563a8795f974425fb26dd0457/Source/Templates/system_stm32h7xx.c#L175-L301
By calling SystemInit() first, we ensure that the external memory is properly configured, allowing the subsequent initialization of the data and BSS sections to proceed without issues.

I hope this clarifies the rationale behind the change. Thank you for your understanding. Allow me to close this issue.

Kind regards,

closed 2024-07-19 by TOUNSTM

cpq, 2024-07-19:
@TOUNSTM 

I don't know what do you mean by `In the SystemInit() API, there are no variables like SystemCoreClock`.
Maybe if you take a look at the CMSIS documentation, https://www.keil.com/pack/doc/CMSIS_Dev/Core/html/group__system__init__gr.html , you'll find that `uint32_t SystemCoreClock` is there, and must be defined by the device code.

Unless you're implementing some other CMSIS, not ARM.

TOUNSTM, 2024-07-19:
Hello @cpq,

We understand your concerns and appreciate your feedback. The initialization of the data and BSS sections before calling SystemInit() is indeed a common practice in embedded systems development. However, in certain scenarios, particularly when using external memory, it is necessary to call SystemInit() before initializing the data and BSS sections to ensure proper memory configuration.
> Nevertheless, as I pointed out elsewhere, some of the ST device CMSIS startup code call SystemInit() before data/BSS init. Some - after. So your argument "This is why SystemInit() must be called before initializing the data and BSS sections" may be valid, but it contradicts your own practice. Like here, for F4 series: https://github.com/STMicroelectronics/cmsis_device_f4/blob/73273c20e85c6c57688f6106485a2e4f2592ddf9/Source/Templates/gcc/startup_stm32f429xx.s#L95-L99

In our specific case, this approach is essential to ensure the correct operation of the system. You can refer to a recent version of the F4 series, where you will see that the issue has been addressed in the same manner.
https://github.com/STMicroelectronics/cmsis_device_f4/blob/cdbad761857acedcdd07ece7939b4cb209ed826a/Source/Templates/gcc/startup_stm32f429xx.s#L63-L100

While your perspective is valid in general terms, it is not applicable to our specific situation due to the need for external memory initialization. We hope this clarifies our approach and the rationale behind it.

Thank you for your understanding.