STMicroelectronics/cmsis-device-h7 #11: Incorrect bit definitions for APB1HFZ1 and FMC_SDCMR registers

carlo-dev-git, 2024-03-25:
**Describe the bug**
For the following devices STM32H723/725/730/733/735 the definition of `DBGMCU_APB1HFZ1_DBG_TIM24_Pos` is incorrect:
`#define DBGMCU_APB1HFZ1_DBG_TIM24_Pos     (24U)`
It should be changed to:
`#define DBGMCU_APB1HFZ1_DBG_TIM24_Pos     (25U)`

For STM32H733 only also `FMC_SDCMR_MODE_2` should be fixed from:
`#define FMC_SDCMR_MODE_2           (0x3UL << FMC_SDCMR_MODE_Pos)                /*!< 0x00000003 */`
to:
`#define FMC_SDCMR_MODE_2           (0x4UL << FMC_SDCMR_MODE_Pos)                /*!< 0x00000004 */`

RJMSTM, 2024-03-29:
Hello @carlo-dev-git,

I agree with you on the first point. I will pass it on to the development team for correction, However, regarding the second point, my stance differs from yours, as detailed in the reference manual rm0468

[image]

The mode 2 setting in the FMC_SDCMR register is correctly positioned at 3, as indicated in the definition found in the tm32h733xx.h file.
[image]
 
Regards,
Rania

RJMSTM, 2024-03-29:
ST Internal Reference: 177832

carlo-dev-git, 2024-03-29:
Hi Rania,
I think you are confusing the bit number with the bit mask:
`(0x3UL << FMC_SDCMR_MODE_Pos)` means `0b0011` which is definitely not MODE[2] bit.
`(0x4UL << FMC_SDCMR_MODE_Pos)` means `0b0100`.

To confirm my statement, please check the header files of other micros.
The following code is taken from `stm32h743xx.h`:
```
/******************  Bit definition for FMC_SDCMR register  *******************/
#define FMC_SDCMR_MODE_Pos         (0U)
#define FMC_SDCMR_MODE_Msk         (0x7UL << FMC_SDCMR_MODE_Pos)               /*!< 0x00000007 */
#define FMC_SDCMR_MODE             FMC_SDCMR_MODE_Msk                          /*!<MODE[2:0] bits (Command mode) */
#define FMC_SDCMR_MODE_0           (0x1UL << FMC_SDCMR_MODE_Pos)                /*!< 0x00000001 */
#define FMC_SDCMR_MODE_1           (0x2UL << FMC_SDCMR_MODE_Pos)                /*!< 0x00000002 */
#define FMC_SDCMR_MODE_2           (0x4UL << FMC_SDCMR_MODE_Pos)                /*!< 0x00000004 */
```

[image]

I hope I have detailed the topic better.

Regards,
Carlo

RJMSTM, 2024-04-23:
Fixed in 88f29645ec7e063bae3d5e58699fc541a81efac6

closed 2024-04-23 by RJMSTM