STMicroelectronics/cmsis-device-h7 #12: D1 Domain Core prescaler array "D1CorePrescTable[16]" non-zero values for values from 0b0100 to 0b111

Fadi-Eid, 2024-04-07:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * STM32H755ZIQ
 * uVision V5.39.0.0

**Describe the bug**
in the _system_stm32h7xx_dualcore_boot_cm4_cm7.c_ file, the function _SystemCoreClockUpdate_ divides the value in the variable _common_system_clock_ based on the four D1CPRE bits of the RCC_D1CFGR register. If the bits are 0b0000 to 0b0111 (0 to 7), sys_clk is not divided, but the first 8 value of the array D1CorePrescTable[16] are {0, 0, 0, 0, 1, 2, 3, 4}, which might cause a problem.
`const  uint8_t D1CorePrescTable[16] = {0, 0, 0, 0, 1, 2, 3, 4, 1, 2, 3, 4, 6, 7, 8, 9};`

`/* Compute SystemClock frequency --------------------------------------------------*/`
 ` tmp = D1CorePrescTable[(RCC->D1CFGR & RCC_D1CFGR_D1CPRE)>> RCC_D1CFGR_D1CPRE_Pos];`
  `/* common_system_clock frequency : CM7 CPU frequency  */`
  `common_system_clock >>= tmp;`

**Screenshots**
[image]

TOUNSTM, 2024-05-28:
ST Internal Reference: 182575

TOUNSTM, 2024-11-26:
Hello @Fadi-Eid ,

The current D1CorePrescTable[16] is implemented to support both 4-bits dividers like D1CPRE and 3-bits dividers such as D1PPRE, D2PPRE2, D2PPRE1. Your proposal is correct for 4-bit dividers, but it is incorrect for 3-bit dividers. Unfortunately, our development teams has no plan to integrate this kind of change request at this time. 

Thank you for your understanding. Please allow me to close this issue.
Kind regards,

ALABSTM, 2026-02-11:
Hi @Fadi-Eid,

Thank you for your report. Actually, this is not an issue but rather a somewhat complex solution that looks tricky at first sight yet works perfectly. Please have a look at [this](https://github.com/STMicroelectronics/cmsis-device-h7/issues/23#issuecomment-3884426195) explanation for details.

With regards,