STMicroelectronics/cmsis-device-h7 #16: Incorrect ETH_DMADSR_RPS Register Definitions

biglben, 2024-09-25:
**Describe the set-up**
STM32H743

**Describe the bug**
For STM32H743 (and probably all other h7 and h5 which have ethernet) the [defines](https://github.com/STMicroelectronics/cmsis_device_h7/blob/13ba9da797a30e1563a8795f974425fb26dd0457/Include/stm32h743xx.h#L8370) for `ETH_DMADSR_RPS_X_POS` are shifted by 4 (and `ETH_DMADSR_RPS_WAITING` has wrong comment).

```c
#define ETH_DMADSR_RPS_Pos                            (8U)
#define ETH_DMADSR_RPS_Msk                            (0xFUL << ETH_DMADSR_RPS_Pos) /*!< 0x00000F00 */
#define ETH_DMADSR_RPS                                ETH_DMADSR_RPS_Msk       /* DMA Channel Receive Process State */
#define ETH_DMADSR_RPS_STOPPED                        (0U)   /* Stopped (Reset or Stop Receive Command issued) */
#define ETH_DMADSR_RPS_FETCHING_Pos                   (12U)
#define ETH_DMADSR_RPS_FETCHING_Msk                   (0x1UL << ETH_DMADSR_RPS_FETCHING_Pos) /*!< 0x00001000 */
#define ETH_DMADSR_RPS_FETCHING                       ETH_DMADSR_RPS_FETCHING_Msk /* Running (Fetching Rx Transfer Descriptor) */
#define ETH_DMADSR_RPS_WAITING_Pos                    (12U)
#define ETH_DMADSR_RPS_WAITING_Msk                    (0x3UL << ETH_DMADSR_RPS_WAITING_Pos) /*!< 0x00003000 */
#define ETH_DMADSR_RPS_WAITING                        ETH_DMADSR_RPS_WAITING_Msk /* Running (Waiting for status) */
#define ETH_DMADSR_RPS_SUSPENDED_Pos                  (14U)
#define ETH_DMADSR_RPS_SUSPENDED_Msk                  (0x1UL << ETH_DMADSR_RPS_SUSPENDED_Pos) /*!< 0x00004000 */
#define ETH_DMADSR_RPS_SUSPENDED                      ETH_DMADSR_RPS_SUSPENDED_Msk /* Suspended (Rx Descriptor Unavailable) */
#define ETH_DMADSR_RPS_CLOSING_Pos                    (12U)
#define ETH_DMADSR_RPS_CLOSING_Msk                    (0x5UL << ETH_DMADSR_RPS_CLOSING_Pos) /*!< 0x00005000 */
#define ETH_DMADSR_RPS_CLOSING                        ETH_DMADSR_RPS_CLOSING_Msk /* Running (Closing the Rx Descriptor) */
#define ETH_DMADSR_RPS_TIMESTAMP_WR_Pos               (13U)
#define ETH_DMADSR_RPS_TIMESTAMP_WR_Msk               (0x3UL << ETH_DMADSR_RPS_TIMESTAMP_WR_Pos) /*!< 0x00006000 */
#define ETH_DMADSR_RPS_TIMESTAMP_WR                   ETH_DMADSR_RPS_TIMESTAMP_WR_Msk /* Timestamp write state */
#define ETH_DMADSR_RPS_TRANSFERRING_Pos               (12U)
#define ETH_DMADSR_RPS_TRANSFERRING_Msk               (0x7UL << ETH_DMADSR_RPS_TRANSFERRING_Pos) /*!< 0x00007000 */
#define ETH_DMADSR_RPS_TRANSFERRING                   ETH_DMADSR_RPS_TRANSFERRING_Msk /* Running (Transferring the received packet data from the Rx buffer to the system memory) */
```

**Screenshots**
RM0433 STM32H742, STM32H743/753 and STM32H750 Rev 8
[image]

ASEHSTM, 2024-09-27:
Hello @biglben,

Thank you for your report. The RPS bits are located from 8 to 11 in the DMADSR register, as shown in the image below, so the values must be shifted by 8 bits.

[image]

With Regards

closed 2024-10-07 by ASEHSTM

biglben, 2024-10-07:
Sorry if i was not clear enough. `ETH_DMADSR_RPS_Pos` is correct but the others (like `ETH_DMADSR_RPS_FETCHING_Pos`, `ETH_DMADSR_RPS_WAITING_Pos` and so on) are wrong. For example `ETH_DMADSR_RPS_FETCHING_Pos` is 12 which is outside of 11:8