STMicroelectronics/cmsis-device-h7 #18: stm32h7 missing register define

dariis, 2025-01-24:
**Describe the set-up**
 * STM32H747BI

**Additional context**

The free running clock bit is missing in the header file.

```diff
 #define QUADSPI_CCR_SIOO_Pos             (28U)
 #define QUADSPI_CCR_SIOO_Msk             (0x1UL << QUADSPI_CCR_SIOO_Pos)         /*!< 0x10000000 */
 #define QUADSPI_CCR_SIOO                 QUADSPI_CCR_SIOO_Msk                    /*!< SIOO: Send Instruction Only Once Mode */
+#define QUADSPI_CCR_FRCM_Pos             (29U)
+#define QUADSPI_CCR_FRCM_Msk             (0x1UL << QUADSPI_CCR_FRCM_Pos)         /*!< 0x20000000 */
+#define QUADSPI_CCR_FRCM                 QUADSPI_CCR_FRCM_Msk                    /* FRCM: Free-running clock mode */
 #define QUADSPI_CCR_DHHC_Pos             (30U)
 #define QUADSPI_CCR_DHHC_Msk             (0x1UL << QUADSPI_CCR_DHHC_Pos)         /*!< 0x40000000 */
 #define QUADSPI_CCR_DHHC                 QUADSPI_CCR_DHHC_Msk                    /*!< DHHC: DDR hold half cycle */
```

KRASTM, 2025-01-27:
ST Internal Reference: 201437

KRASTM, 2025-02-13:
Fixed in Commit: https://github.com/STMicroelectronics/cmsis-device-h7/commit/cf3743a6e1a94e381ab15e0899f555f518099567

closed 2025-02-13 by KRASTM