STMicroelectronics/cmsis-device-h7 #20: Is it planned to remove old code and macros?

KoynovStas, 2025-03-16:
CMSIS device files (headers) are very large, and changes are periodically made like: https://github.com/STMicroelectronics/cmsis-device-h7/commit/e4d57d6edc39f35aced298732829fc2a5e6a8a71

This brings each time a section with `Legacy xxx`
This is almost doubling the code, old + new.

Is it planned to remove the old code or is it left for compatibility and will never be removed?.
If for compatibility, then if it is with us forever, why introduce new names for the same thing if the old code will 100% not be removed?

If at least some plans for a large-scale refactoring that will remove old definitions?

I ask because I would like to have synchronized versions of these files with the official repository.

But if this refactoring is not planned, we will plan refactoring in our repository, because old/new code with a many of developers leads to an increase in entropy.

Thank you.

ASEHSTM, 2025-03-24:
Hello @KoynovStas,

The legacy code is maintained for compatibility reasons and will not be removed. 
Developers have the discretion to choose which version to use. However, we strongly recommend utilizing the new code for any new development.

Thank you.
With Regards,

KoynovStas, 2025-03-24:
Thank you for the official response.

closed 2025-03-24 by KoynovStas